chandrawi/LoRaRF-Arduino #1: Is there some Channel Activity Detection?

bobricius, 2021-03-05:
Hi,  I searching for alternative library to https://github.com/sandeepmistry/arduino-LoRa
Is there some Channel Activity Detection or something to tell my if is channel free?
Is supported SamD21?

Thanks

chandrawi, 2021-03-15:
Halo Peter Misenko,

I made LoRa library for Arduino platform. I assume you already visit the
library I publish in github. This is the github link of the LoRa arduino
llibrary you probably seen
https://github.com/chandrawi/LoRaRF-Arduino

I also made LoRa library using python for SBC like Raspberry pi. If you
interested in developing LoRa device or gateway using raspberry pi, you can
check library in following link
https://github.com/chandrawi/LoRaRF-Python

Both library support for SX126x LoRa chip family including SX1261, SX1262
and SX1268. But unfortunately, both library not yet support for Channel
Activity Detection. The LoRa arduino library should be support for arduino
Zero with SAMD21 processor, but I haven't check it. I only check on arduino
uno and pro mini 3.3V and work properly using E22400M22S,  E22900M22S,
E22400M22S, and E22900M30S Ebyte LoRa module. If you plan to use arduino
zero you should set SPI setting to use SPI1 instead SPI0 using setSPI()
method
```
#include <SX126x.h>
SX126x LoRa;
setSPI(SPI1);
```

I have plan to add CAD feature for both LoRa library in near feature. I
rarely check email, if you want immediate response you chat me
on +6285321251627

Regards,
Chandra Wijaya

chandrawi, 2021-03-15:
Halo Peter Misenko,

The library support for SX126x LoRa chip family including SX1261, SX1262 and SX1268. But unfortunately, the library not yet support for Channel Activity Detection. The LoRa arduino library should be support for arduino Zero with SAMD21 processor, but I haven't check it. I only check on arduino uno and pro mini 3.3V and work properly using E22400M22S,  E22900M22S, E22400M22S, and E22900M30S Ebyte LoRa module. If you plan to use arduino zero you should set SPI setting to use SPI1 instead SPI0 using setSPI() method
```
#include <SX126x.h>
SX126x LoRa;
setSPI(SPI1);
```

I also made LoRa library using python for SBC like Raspberry pi. If you
interested in developing LoRa device or gateway using raspberry pi, you can
check library in following link
https://github.com/chandrawi/LoRaRF-Python

I have plan to add CAD feature for both LoRa library in near feature. I
rarely check email, if you want immediate response you chat me
on +6285321251627

Regards,
Chandra Wijaya

chandrawi, 2021-03-15:
Hallo Peter Misenko,

The library support for SX126x LoRa chip family including SX1261, SX1262 and SX1268. But unfortunately, the library not yet support for Channel Activity Detection. I have plan to add CAD feature for the library in the future.
The LoRa arduino library should be support for arduino Zero with SAMD21 processor, but I haven't check it. I only check on arduino uno and pro mini 3.3V and work properly using E22400M22S, E22900M22S, E22400M22S, and E22900M30S Ebyte LoRa module. If you plan to use arduino zero you should set SPI setting to use SPI1 instead SPI0 using setSPI() method
```
#include <SX126x.h>
SX126x LoRa;
setSPI(SPI1);
```

Regards,
Chandra Wijaya

closed 2022-01-20 by chandrawi