chandrawi/LoRaRF-Arduino #3: rp2040-sx1278

01Space, 2022-03-14:
Can you make it support rp2040 development board?

chandrawi, 2022-03-14:
This LoRa library built for Arduino framework. If you using Arduino, this library will worked. You should check that Arduino has RP2040 development board.

https://docs.arduino.cc/hardware/nano-rp2040-connect

Please note that this library has dependency with some Arduino core library including SPI library. So, clearly this library will not working outside Arduino development environment.

chandrawi, 2022-03-14:
Update, I have check Arduino RP2040 core library [here](https://github.com/earlephilhower/arduino-pico).

I think using this core library, LoRa library should work. But `onReceive()` and `onTransmit()` methods will not work because `attachInterrupt()` is not implemented in the RP2040 SPI library. It is described [here](https://arduino-pico.readthedocs.io/en/latest/spi.html).

Currently I don't have Raspberry pi pico or any RP2040 board. So, I can't test it yet.

01Space, 2022-03-14:
thank you!  https://github.com/earlephilhower/arduino-pico  Arduino can support rp2040, but the current Lora libraries do not support rp2040 (raspberry PI Pico)，I designed a Lora + 2040 + LCD development board

chandrawi, 2022-03-14:
Can you point out at which point this LoRa library does not support rp2040?

Some error message while you trying upload Arduino sketch, may be? So I can help to fix compatibility issue with rp2040.

01Space, 2022-03-14:
Arduino:1.8.16 (Windows 10), ："Raspberry Pi Pico, 2MB (no FS), 125 MHz, Small (-Os) (standard), Disabled, Disabled, None, Pico SDK"

C:\Users\86186\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.12.0\libraries\LoRaRF\src\SX126x_API.cpp: In static member function 'static bool SX126x_API::reset(int8_t)':

C:\Users\86186\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\1.12.0\libraries\LoRaRF\src\SX126x_API.cpp:29:1: error: no return statement in function returning non-void [-Werror=return-type]

   29 | }

      | ^

cc1plus.exe: some warnings being treated as errors

exit status 1

thank you! I choose Pico to compile, but I can't compile

chandrawi, 2022-03-14:
Try to modify function type from `bool` to `void` in SX126x_API.cpp and SX126x_API.h files located at LoRaRF library folder.

Change in SX126x_API.cpp
```cpp
bool SX126x_API::reset(int8_t reset)
{
    pinMode(reset, OUTPUT);
    digitalWrite(reset, LOW);
    delay(1);
    digitalWrite(reset, HIGH);
}
```

to

```cpp
void SX126x_API::reset(int8_t reset)
{
    pinMode(reset, OUTPUT);
    digitalWrite(reset, LOW);
    delay(1);
    digitalWrite(reset, HIGH);
}
```

and in SX126x_API.h 

```cpp
static bool reset(int8_t reset);
```
 to

```cpp
static void reset(int8_t reset);
```

01Space, 2022-03-14:
thank you! I just used https://github.com/jgromes/RadioLib This library can accept data, but it must be connected to sx1278 dio1 pin

closed 2022-03-15 by 01Space