chandrawi/LoRaRF-Arduino #4: Heltec CubeCell 6502s 'SX126x' does not name a type error

jc1195, 2022-04-09:
I keep having the error 'SX126x' does not name a type. The board uses a ASR6502mcu. I've tried the examples you have provided and keep getting the same error as well. I set the pins according to the pinout https://resource.heltec.cn/download/CubeCell/HTCC-AB02S/HTCC-AB02S_PinoutDiagram.pdf.
int8_t nssPin = 28, resetPin = 35, busyPin = 34, irqPin = 33, txenPin = 18, rxenPin = 17; I was hoping that you could help out. I'm still new to this. I'm just trying to make it send a message with GPS coordinates using lora.

chandrawi, 2022-04-09:
It is most likely because the library is not installed correctly. Try to check if your Arduino library folder has LoRaRF folder and there is SX126x.h and SX126x.cpp inside /path-to-arduino-library/LoRaRF-Arduino/src folder. 

If you sure there is those files. Try to uninstall then reinstall library again via Arduino library manager.

jc1195, 2022-04-09:
> It is most likely because the library is not installed correctly. Try to check if your Arduino library folder has LoRaRF folder and there is SX126x.h and SX126x.cpp inside [...]

I just tried doing that and I still have the issue of it saying "SX126x does not name a type". I'm also using the visual micro extension in visual studio. However, I tried using the same code in Arduino IDE and throws the same error there as well.

chandrawi, 2022-04-10:
I have install CubeCell arduino board and try to compile SX126x_LoRa_receiver and SX126x_LoRa_transmitter examples. I know why "SX126x does not name a type" error happen. It is because conflict between CubeCell arduino core and LoRaRF library.  CubeCell arduino core has `sx126x.h` and `sx126x.cpp` files and LoRaRF has `SX126x.h` and `SX126x.cpp`.

After I delete sx126x.h and sx126.cpp files in `C:\Users\<your-windows-username>\AppData\Local\Arduino15\packages\CubeCell\hardware\CubeCell\1.4.0\cores\asr650x\device` the error no longer happen. This action may break CubeCell LoRa library. So you must choose between using CubeCell LoRa library or LoRaRF library.

jc1195, 2022-04-10:
Oh thank you for that! I didn't even think about that. I just changed the name in one of the headers and now it compiles. Now I just need to figure out the correct pins 😂

closed 2022-04-22 by chandrawi