chandrawi/LoRaRF-Arduino #5: Esp8266 ESP32 cannot use lorarf Library

yaniot, 2022-07-03:
When using lorarf, use the SPI pins of esp8266 and esp32. When uploading sketches, the esp8266 door opener is turned on, esp32 has no response, and Arduino nano can be used.
------------------------------------------------------------------------------
Operating environment:
Esp8266 uses nodemcu 0.9
Esp32 uses esp32 dev module
Arduino IDE     1.8.19
LoRaRF library 2.0.0

LSC862, 2022-07-18:
Have you solved the problem now?I tried to add print to.cpp and.h, and found that the function in app.h does not seem to work, it seems that there is no write and read function in SPI, _write/_readBytes mentioned in app.h seems not to be the method in spi.h. I don't know if I'm right. If you have solved it, could you please send a demo to me for study? Thank you for your

chandrawi, 2022-09-03:
I try the suggestion by changing the default SPI speed and it is works. I just upload new 2.1.0 version. Now LoRaRF library works on ESP32 and ESP8266.

closed 2022-09-03 by chandrawi

chandrawi, 2022-10-11:
Dear Yanqisui,

There is issue using LoRaRF library with ESP32 and ESP8266. The issue is
SPI bus Arduino library compatibility. I am working around this issue.
Hopefully LoRaRF library will support ESP32 soon.

The microcontroller supported by LoRaRF library are ATmega based Arduino
(Uno, nano, pro mini) and STM32duino.

Regards,