chandrawi/LoRaRF-Arduino #6: GetDeviceErrors incorrect read

ramonverde, 2022-10-05:
I'm getting incorrect reads from GetDeviceErrors with LLCC68.
The library is sending 0x17 byte, as expected, but it must read 3 bytes from the response, but in sx126x_getDeviceErrors it is only reading 2 of them.
This creates an offset on SPI communication, getting erroneous reads from LLCC68 since then (1 byte offset).
In sx126x_getDeviceErrors, the line: sx126x_transfer(0x17, buf, 2);
must be changed for: sx126x_transfer(0x17, buf, 3);

chandrawi, 2022-10-06:
Yes, it is a mistake. I have checked both SX1261_2 and LLCC68 datasheet and found that GetDeviceErrors command has two bytes data like you said.
Can you just create a pull request for this fix?

ramonverde, 2022-10-07:
Created pull request #7

ramonverde, 2022-10-09:
Fixed with pull request #7

closed 2022-10-09 by ramonverde