chandrawi/LoRaRF-Arduino #10: Connecting and working with STM32

GeorgyDemyanov, 2022-10-15:
Hello, sorry to bother you, I have a question how to connect LoRa SX1268 to STM32 (nucleo - F429ZI). Could you give me some advice on hardware connection? And what needs to be added to the firmware to make it work with STM32?

chandrawi, 2022-10-15:
For using this library to make a STM32 microcontroller as host of LoRa module, you need to install the [STM32duino core](https://github.com/stm32duino/wiki/wiki/Getting-Started) and [STM32 cube programmer](https://www.st.com/en/development-tools/stm32cubeprog.html). Only those two software tools are needed besides Arduino IDE, no other software tools or firmware are needed. If you sure you have installed all those software correctly then you can upload an Arduino sketch which include this library to your STM32 microcontroller. Unfortunately, I have not tried using nucleo `F429ZI`. But I have tried upload the sketch to `STM32F103` (blue pill), `STM32L151CCU6`, and `STM32L051CBT6`. The sketch works perfectly as I expected on those microcontrollers.

SX1268 is a chinese version of SX126x LoRa family. I have not tried this LoRa chip, but as far I know all the op commands and registry is the same as SX1262, so I think using `SX126x` header will works with this chip.

You should try to upload `SX126x_LoRa_transmitter` and `SX126x_LoRa_receiver` example sketches to two microcontrollers. Then check if the receiver receive 'HeLoRa World!' message from the transmitter. The connection I use with STM32F103 blue pill are like in the following.

| SX126x | STM32F103 |
| :------------: | :-------------:|
| VCC | 3V3|
| GND | GND |
| SCK | PA5 |
| MISO | PA6 |
| MOSI | PA7 |
| NSS | PA4 | 
| RESET | RESET | 
| BUSY | PA4 |

You can change SPI pins (`SCK`, `MISO`, `MOSI`) to meet SPI pins in `STM32F429ZI`. For `RESET` and `BUSY` pins you can connect to any GPIO pins.

After you make the connections, change the pin definition in `SX126x_LoRa_transmitter` and `SX126x_LoRa_receiver` examples. For examples, using the above connections the pin definition would look like this.

```cpp
  SPIClass SPI_2(PA7, PA6, PA5);
  LoRa.setSPI(SPI_2, 16000000);

  int8_t nssPin = PA4, resetPin = PA3, busyPin = PA2, irqPin = PB11, txenPin = -1, rxenPin = -1;
  if (!LoRa.begin(nssPin, resetPin, busyPin, irqPin, txenPin, rxenPin)){
    Serial.println("Something wrong, can't begin LoRa radio");
    while(1);
  }
```

GeorgyDemyanov, 2022-10-15:
Thanks for your reply.  I will try soon

GeorgyDemyanov, 2022-10-17:
Okay, it worked, the message is sent, but the method LoRa.wait(), works for a very long time, the transmission time "123" takes about 2 minutes. Is this a modulation issue or am I doing something wrong?
Also, another question arose. Principle of operation LoRaRF Is it the same for Arduino and Python? Because I have a device Raspberry pi, and it does not receive a signal from STM32. Is this a code issue, or mine? Thanks in advance! Thanks in advance!

chandrawi, 2022-10-18:
`LoRa.wait()` method is used to make sure a transmit or receive process finished before doing something else. For example, if you want to put LoRa to sleep mode after transmitting message you must place `LoRa.wait()` before `LoRa.sleep()` so modulation process is not interrupted. The `LoRa.endPacket()` and `LoRa.request()` method only configuring LoRa to transmit and receive a packet. Those methods don't wait transmit or receive process to finish.

2 minutes is too long for a LoRa packet transmission. The transmit time is ranged between tens millisecond to a few seconds depending on spreading factor and bandwidth setting. Transmit time in `SX126x_LoRa_transmitter` example should be around 65 ms. Try to do following to fix this problem.

- Deactivate interrupt operation mode by change `irqPin` to -1 instead of PB11 or any pin you set. The interrupt signal detection may be broken, so leave DIO1 pin unconnected. LoRa can run without DIO1 connection.
- Keep or delete `LoRa.setDio3TcxoCtrl(dio3Voltage, tcxoDelay)` line. Some LoRa module cannot work properly without this setting, and other cannot work using this setting.
- Keep or delete `LoRa.setDio2RfSwitch(true)` line. If your LoRa module datasheet mentioning to use `TXEN` and `RXEN` pins delete this line, add connection to those pins, and change `txenPin` and `rxenPin` to the pin you connected. If your LoRa module doesn't have `TXEN` and `RXEN` pin keep this line.

The principle of operation LoRaRF Arduino library is same with [LoRaRF Python library](https://github.com/chandrawi/LoRaRF-Python). I have tried and successfully transmitting LoRa packet from STM32F103 to raspberry pi and vice versa.

GeorgyDemyanov, 2022-10-18:
Thank you so much for your invaluable advice!