chandrawi/LoRaRF-Arduino #21: Why does the module receive garbled data sent by itself when I merge the sending and receiving codes together

ShiPC123, 2023-12-12:
I can use LLCC68 module to send data normally, but if I combine the receiving and sending code together, after the sending code is executed, the serial port will print the received garbled data

[image]
[image]
[image]

ShiPC123, 2023-12-12:
How can I use this code so that I can implement the sending and receiving function with one code

i-am-bigmike, 2024-01-06:
Today I was doing a similar test, and I’ve encountered the same problem.

Altought I should investigate better, I think the problem is the variable “_payloadTxRx”.

After a send packet command, _payloadTxRx contains the length of the data sent.
Then, when you call LoRa.avail(), it returns the value of the same variable, still containing number of bytes sent, so the program thinks some data has been received.

I solved resetting this value when switching from TX mode to RX mode modifying the LoRa.listen function.
By the way, I don’t see the “listen” or “request” command in your source, while I think it’s necessary to switch to RX mode.

If you don’t want to modify the library source, you could add:
LoRa.purge(LoRa.available());
just immediately after LoRa.wait();

i-am-bigmike, 2024-01-22:
I found another problem that drove me crazy.
When switching from RX (without receiving data) to TX, the data sent often contains only garbage characters. 
After a long debugging session, I identified the problem. According to the LLCC68 datasheet, to switch to TX mode, the first step is: _“if not in STDBY_RC mode, then switch to this mode using the SetStandby(...) command.”_
So I added these lines in the SX126x::beginPacket function just before everything:
```c
    /* From datasheet DS_LLCC68 - "Circuit Configuration for Basic Tx Operation" :
        1.If not in STDBY_RC mode, then go to this mode
    */
    if (getMode() != SX126X_STATUS_MODE_STDBY_RC) {
      sx126x_setStandby(SX126X_STANDBY_RC);
    }
```