chandrawi/LoRaRF-Arduino #29: Compile LD error...\LoRaRF\src/SX126x.cpp:128: undefined reference to `vtable for BaseLoRa'

kigik, 2024-07-10:
I have trouble builing the example SX126x_LoRa_receiver_listen, with ESP32 Dev  Module, esp32 3.0.0 version hardware
but fail, and it outputs:
...\libraries\LoRaRF\SX126x.cpp.o:(.literal._ZN6SX126xC2Ev+0x0): undefined reference to `vtable for BaseLoRa'
collect2.exe: error: ld returned 1 exit status
Is there something need modify with this example.

LithosphereRocketry, 2025-03-13:
I also have this issue on STM32duino, although only in debug mode and not release mode.

LithosphereRocketry, 2025-03-13:
I think I've located the issue - the BaseLora class is using virtual functions in an unsafe way that some compilers & compiler settings sweep under the rug but others are pickier about. It can be fixed by setting the functions in that class to pure-virtual like in this diff:
https://github.com/LithosphereRocketry/LoRaRF-Arduino/commit/ddbb43d6cb8292aaa44fb0a12a46640694cb09c4