chandrawi/LoRaRF-Arduino #31: Garbled packets on Heltec wifi lora stick v3

chedim, 2024-09-02:
Boards: https://heltec.org/project/wireless-stick-v3/
Trying to send "hello" messages between the two of these and instead receiving garbled (seemingly not random as they repeat) characters:

```
loragate/debug [D][lora_mqtt_bridge.sensor:101]: received packet: ?@m$
loragate/debug [D][lora_mqtt_bridge.sensor:070]: Packet size: 5
loragate/debug [D][lora_mqtt_bridge.sensor:101]: received packet: z?H?
loragate/debug [D][lora_mqtt_bridge.sensor:070]: Packet size: 5
loragate/debug [D][lora_mqtt_bridge.sensor:101]: received packet: z?H?
loragate/debug [D][lora_mqtt_bridge.sensor:070]: Packet size: 5
loragate/debug [D][lora_mqtt_bridge.sensor:101]: received packet: ???s
loragate/debug [D][lora_mqtt_bridge.sensor:070]: Packet size: 5
loragate/debug [D][lora_mqtt_bridge.sensor:101]: received packet: ???s
loragate/debug [D][lora_mqtt_bridge.sensor:070]: Packet size: 5
loragate/debug [D][lora_mqtt_bridge.sensor:101]: received packet: ???s
```

Transmitter initialization code:
```
	    if (!LoRa.begin(cs_pin, reset_pin, busy_pin, dio0_pin, -1, -1)) {
		    ESP_LOGE(TAG, "Failed to setup LoRa, mode: %d, error: %d", LoRa.getMode(), LoRa.getError());
		    this->mark_failed();
		    return;
	    }

		uint8_t dio3Voltage = SX126X_DIO3_OUTPUT_1_8;
		uint32_t tcxoDelay = SX126X_TCXO_DELAY_10;
		LoRa.setDio3TcxoCtrl(dio3Voltage, tcxoDelay);
		LoRa.setTxPower(17, SX126X_TX_POWER_SX1262);                        // TX power +17 dBm for SX1262

            LoRa.setSyncWord(_sync);
	    LoRa.setFrequency(_frequency);
	    ESP_LOGI(TAG, "LoRa Modulation: spread=%d, bandwidth=%d, coding=%d", _spread, _bandwidth, _coding);
	    LoRa.setLoRaModulation(_spread, _bandwidth, _coding, false);

		uint8_t headerType = SX126X_HEADER_EXPLICIT;                        // Explicit header mode
		uint16_t preambleLength = 12;                                       // Set preamble length to 12
		uint8_t payloadLength = 255;                                         // Initialize payloadLength to 15
		bool crcType = false;                                                // Set CRC enable
		LoRa.setLoRaPacket(headerType, preambleLength, payloadLength, crcType);
```
Transmitter code:
```
            LoRa.beginPacket();
	    LoRa.write("hello", 5);

            if (LoRa.endPacket(20)) 
		    ESP_LOGI(TAG, "sent a packet");
	    else {
		    ESP_LOGW(TAG, "failed to send a packet");
		    return;
	    }

	    LoRa.wait(200);
```
---
Receiver initialization code:
```
	    if (!LoRa.begin(cs_pin, reset_pin, busy_pin, dio0_pin, -1, -1)) {
		    ESP_LOGE(TAG, "Failed to setup LoRa, mode: %d, error: %d", LoRa.getMode(), LoRa.getError());
		    this->mark_failed();
		    return;
	    }

	      // Configure TCXO or XTAL used in RF module
		  uint8_t dio3Voltage = SX126X_DIO3_OUTPUT_1_8;
		  uint32_t tcxoDelay = SX126X_TCXO_DELAY_10;
		  LoRa.setDio3TcxoCtrl(dio3Voltage, tcxoDelay);
	      // Set RX gain to boosted gain
		  LoRa.setRxGain(SX126X_RX_GAIN_BOOSTED);

	    LoRa.setFrequency(_frequency);
	    ESP_LOGI(TAG, "LoRa Modulation: spread=%d, bandwidth=%d, coding=%d", _spread, _bandwidth, _coding);
	    LoRa.setLoRaModulation(_spread, _bandwidth, _coding, false);
	    LoRa.setSyncWord(_sync);

	    uint8_t headerType = SX126X_HEADER_EXPLICIT;                        // Explicit header mode
	    uint16_t preambleLength = 12;                                       // Set preamble length to 12
	    uint8_t payloadLength = 255;                                         // Initialize payloadLength to 15
	    bool crcType = false;                                                // Set CRC enable
	    LoRa.setLoRaPacket(headerType, preambleLength, payloadLength, crcType);

	    if (!LoRa.request(SX126X_RX_CONTINUOUS)) {
		    ESP_LOGE(TAG, "Failed to request continuous listening");
		    this->mark_failed();
		    return;
	    }
	    ESP_LOGI(TAG, "Initialized LoRa");
 ```

Receiver code:
```
	    size_t _packet_size = LoRa.available();
	    if (_packet_size == 0) {
		    return;
	    }
		ESP_LOGD(TAG, "Packet size: %d", _packet_size);
                char received_string[_packet_size];

                // received data
		for(int i = 0; i < _packet_size; i++) {
			received_string[i] = LoRa.read();
		}

                ESP_LOGI(TAG, "received packet: %s", received_string);
```

Spent the whole day trying to figure out waidw. Please help, thank you!

m4mans, 2024-10-29:
@chedim Did you find the issue? 

Getting similar packets with LLCC68.

chedim, 2025-01-13:
No, I'm still receiving all sorts of mess :(